brendan-w/python-OBD #233: Get voltage from adapter while connected to vehicle

rack201, 2021-12-25:
With no supported PID for battery voltage on my car, I'm pulling it from my adapter via ATRV in my main OBD loop.  Not a programmer and never used python, so wonder if there is a more elegant way to do this.  Tried to build a custom command but as it's not an OBD response I couldn't make it work.

```
def obdStuff():
    tic=time.time()
    c = obd.OBD("/dev/rfcomm0", baudrate=115200)
    time.sleep(1)
    ser = Serial("/dev/rfcomm0", 115200, timeout=0.5)
    while True:
        toc=time.time()
        try:
            if toc-tic > 5:
                ser.write(b'ATRV\r')
                data_in = ser.read(ser.in_waiting)
                if str(data_in, 'UTF8') !="":
                    #print("V is:", str(data_in, 'UTF8'))
                    client.publish("Voltage", str(data_in, 'UTF8'))
                    tic = time.time()
                    ...
```

incense, 2022-01-18:
The following might be what you have been looking for:
cmd = obd.commands.ELM_VOLTAGE
(found in commands.py).
Haven't it tested though.

Mohit-Agarwa1, 2025-05-26:
> obd.commands.ELM_VOLTAGE

I’ve tried this in my own car and have gotten a solid 14V, as expected.