brendan-w/python-OBD #248: Py-OBD with macOsMonterey (v12.3.1) - MacBook Pro 15

marcusx87, 2022-06-04:
It does not work with macOS Monterey v.12.3.1 (MacBook Pro 15) . . . . . Why ?

Stealthmate, 2023-02-05:
I just tried using this library with a Bluetooth ELM327 dongle today on a Mac (Monterey 12.4) and a Linux laptop (Mint 21). Mac didn't work but Linux did.

I can only speculate, but I think the reason is that serial ports over Bluetooth are either broken on MacOS, or I am not smart enough to be able to set up the port. My Bluetooth dongle showed up as `/dev/tty.OBDII` or `/dev/cu.OBDII`, but specifying that when connecting didn't work, because the `python-OBD` would time out waiting for a response.

On the other hand, on Linux it was easy. Mint offers the option to open a serial port for the Bluetooth device (/`dev/rfcomm01` or something), `python-OBD` automatically sees that and I managed to read real time data from my car.

marcusx87, 2023-02-05:
I don’t know if there is any trick to bypass this problem.

Kiran-G1, 2023-09-11:
I'm facing the same issue on Mac Ventura OS.

Bluetooth connects seamlessly using the UI and I can see the serial ports /dev/tty.OBDII.

But the Python-OBD doesn't connect. 
Sometimes it says resource busy and sometimes it won't detect.