brendan-w/python-OBD #252: Python 3.11 and collections

NFSpeedy, 2022-11-26:
Hi,

I am using Python 3.11 and when importing obd, I get an exception:

```
ImportError                               Traceback (most recent call last)
File ~/.local/share/virtualenvs/pyserial_rasp-JGlkwlhx/lib/python3.11/site-packages/pint/compat/__init__.py:64
     63 try:
---> 64     from collections import Chainmap
     65 except ImportError:

ImportError: cannot import name 'Chainmap' from 'collections' (/Users/nfspeedy/.pyenv/versions/3.11.0/lib/python3.11/collections/__init__.py)

During handling of the above exception, another exception occurred:

ImportError                               Traceback (most recent call last)
Cell In [11], line 1
----> 1 import obd
      3 connection = obd.OBD() # auto-connects to USB or RF port
      5 cmd = obd.commands.SPEED # select an OBD command (sensor)

File ~/.local/share/virtualenvs/pyserial_rasp-JGlkwlhx/lib/python3.11/site-packages/obd/__init__.py:41
     10 ########################################################################
     11 #                                                                      #
     12 # python-OBD: A python OBD-II serial module derived from pyobd         #
   (...)
     37 #                                                                      #
     38 ########################################################################
     40 from .__version__ import __version__
...
---> 16 from collections import MutableMapping
     17 if sys.version_info < (3, 0):
     18     from thread import get_ident

ImportError: cannot import name 'MutableMapping' from 'collections' (/Users/nfspeedy/.pyenv/versions/3.11.0/lib/python3.11/collections/__init__.py)
```

Is Py3.11 going to be supported?

Ch1x3, 2022-12-16:
in files : 
` /.local/lib/python3.11/site-packages/pint/util.py
 /.local/lib/python3.11/site-packages/pint/compat/chainmap.py
`
change : 
` from collections  import Mapping
`
to
`
from collections.abc import Mapping
`

Mrc527, 2023-01-23:
> in files :
> `/.local/lib/python3.11/site-packages/pint/util.py`
> and
> ` /.local/lib/python3.11/site-packages/pint/compat/chainmap.py`
> change :
> `from collections import Mapping`
> to
> `from collections.abc import Mapping`

Actually I still have the issue even after changing this.

Maybe we also need to do something about the 'MutableMapping'?

alistair23, 2023-01-23:
Can you send a PR with those changes?

Mrc527, 2023-01-24:
> Can you send a PR with those changes?

I would. But that has to be changed in the "Pint" package which is not part of this repo right?
I see there is a "Pint" version 0.20.1 which is newer than the currently used 19 version. Maybe upgrading would solve the issue?

alistair23, 2023-01-24:
Ah, whoops. I didn't look closely enough.

It seems like this has been fixed in `pint` [awhile ago](https://github.com/hgrecco/pint/pull/745). Are you using an old version?

Mrc527, 2023-01-24:
I simply did `pip3 install obd` and got that version (19) of `pint`.

I would expect this version to be set in obd, isn't it?

alistair23, 2023-01-24:
I've submitted a PR to update the tests to test Python 3.11, they seem to pass there: https://github.com/brendan-w/python-OBD/pull/255

Stealthmate, 2023-02-05:
I just bumped into this issue today, trying to run via python 3.10. I tried using the branch from #255 instead and it seems to be working!

alistair23, 2023-02-05:
Are you able to review the PR then? That will allow me to merge it

CaptainKatsuura, 2023-05-30:
On python 3.10.9 I receive the same import error due to the deprecation of classes in collections by using the standard pip install obd.

lamemakes, 2023-05-30:
what version of pint are you using specifically?

alistair23, 2023-06-01:
Can you include details on what fails with the latest from git? Let's get that fixed then we can focus on a release

CaptainKatsuura, 2023-06-02:
@alistair23 @lamemakes 

Apologies for the delayed response. I was using pint 0.20.1.

I initially worked around this by reinstalling obd through the standard pip install obd. Then i manually edited pint as suggested by @Ch1x3.

I tried to reproduce the error via the github installation again but found that it now works. This is strange as the dependencies/default setup is the same.

For reference the steps I took are:

`pip install git+https://github.com/brendan-w/python-OBD`

I successfully installed obd-0.7.1 pint-0.20.1 and successfully imported it.

NitsujY, 2023-07-06:
> @alistair23 @lamemakes
>
> Apologies for the delayed response. I was using pint 0.20.1.
>
> I initially worked around this by reinstalling obd through the standard pip install obd. Then i manually edited [...]

I was wondering if it would be possible to have a new release with this patch included? It's been found to be very helpful in fixing the problem in Python 3.11.

alistair23, 2023-07-08:
I have asked @brendan-w to push a release

brendan-w, 2023-07-10:
released as `0.7.2`