brendan-w/python-OBD #258: Some classes in the collections library are deprecated

ortunoa, 2023-02-15:
Obd uses MutableMapping and Mapping, which are in util.py and chainmap.py

"from collections import Mapping" needs to be changed to "from collections.abc import Mapping" on util.py
"from collections import MutableMapping" needs to be changed to "from collections.abc import MutableMapping" on chainmap.py

radonc0de, 2023-02-19:
is there a known workaround for now? as of now, importing obd fails.

lamemakes, 2023-05-18:
How did you install the package? There's a discrepancy between what's in the repo and what's in pypi (pip). For me, the pint issues were revolved via: `python3 -m pip install git+https://github.com/brendan-w/python-OBD#egg=obd` instead of the suggested `python3 -m pip install obd`

Not sure what the state of the project is now, but some docs updates & a new package release would be sweet.

CaptainKatsuura, 2023-05-30:
> How did you install the package? There's a discrepancy between what's in the repo and what's in pypi (pip). For me, the pint issues were revolved via: `python3 -m pip install git+https://github.com/brendan-w/python-OBD#egg=obd` instead of the suggested `python3 -m pip install obd`. See #252 for more.
> 
> Not sure what the state of the project is now, but some docs updates & a new package release would be sweet.

in trying this, importing OBD failed due to the pint requirement. Detailed it in #252