brendan-w/python-OBD #262: b'0100': Supported PIDs [01-20] did not receive any acceptable messages

JKondro, 2023-06-08:
I'm using a ELM 327 V1.5 OBD2 adapter/cable and I'm unable to send commands (eg. engine coolant temp).

The debug log shows "No valid data for PID listing command: b'0100': Supported PIDs [01-20]". From what I can tell, based on my code, is that I've successfully made a connection to the vehicle ('09 Ford Fusion), yet I'm unable to send a command.

I've ensured that the ELM 327's drivers are up to date. And I've ensured that I've upgraded obd (w/ pip).

I'm managed to use python-obd with my ELM 327 on a Mac laptop about a year and half ago. But now I can't seem to get it to 'fully' connect on my Lenovo laptop with Windows 10.

What am I missing?

Here is my code:
```
        ports = obd.scan_serial()
        print (ports)
        obd.logger.setLevel(obd.logging.DEBUG)
        
        # 2009 Ford Fusion
        vehicle = obd.OBD(portstr=ports[0], baudrate=38400, protocol="1", fast=False, timeout=30, check_voltage=False) 

        status = vehicle.is_connected() 
        print(status)

        temp = vehicle.query(obd.commands.COOLANT_TEMP) #Celsius
        print(temp.value)

        vehicle.close() # Ends Connection to Vehicle
```

Here is the debug log report:

>           [obd.obd] ======================= python-OBD (v0.7.1) =======================
>           [obd.obd] Explicit port defined
>           [obd.elm327] Initializing ELM327: PORT=\.\COM3 BAUD=38400 PROTOCOL=1
>           [obd.elm327] write: b'ATZ\r'
>           [obd.elm327] wait: 1 seconds
>           [obd.elm327] read: b'ATZ\r\r\rELM327 v1.5\r\r>'
>           [obd.elm327] write: b'ATE0\r'
>           [obd.elm327] read: b'ATE0\rOK\r\r>'
>           [obd.elm327] write: b'ATH1\r'
>           [obd.elm327] read: b'OK\r\r>'
>           [obd.elm327] write: b'ATL0\r'
>           [obd.elm327] read: b'OK\r\r>'
>           [obd.elm327] write: b'AT RV\r'
>           [obd.elm327] read: b'11.8V\r\r>'
>           [obd.elm327] write: b'ATTP1\r'
>           [obd.elm327] read: b'OK\r\r>'
>           [obd.elm327] write: b'0100\r'
>           [obd.elm327] read: b'NO DATA\r\r>'
>           [obd.elm327] Connected Successfully: PORT=\.\COM3 BAUD=38400 PROTOCOL=1
>           [obd.obd] querying for supported commands
>           [obd.obd] Sending command: b'0100': Supported PIDs [01-20]
>           [obd.elm327] write: b'0100\r'
>           [obd.elm327] read: b'NO DATA\r\r>'
>           [obd.OBDCommand] b'0100': Supported PIDs [01-20] did not receive any acceptable messages
>           [obd.obd] No valid data for PID listing command: b'0100': Supported PIDs [01-20]
>           [obd.obd] finished querying with 7 commands supported
>           [obd.obd] ===================================================================
>           True
>           [obd.obd] Sending command: b'0105': Engine Coolant Temperature
>           [obd.elm327] write: b'0105\r'
>           [obd.elm327] read: b'NO DATA\r\r>'
>           [obd.OBDCommand] b'0105': Engine Coolant Temperature did not receive any acceptable messages
>           None
>           [obd.obd] Closing connection
>           [obd.elm327] closing port
>           [obd.elm327] write: b'ATZ\r'

csurf, 2023-06-16:
I'm dealing with the same issue, but your situation is a bit different. In your debug output, you're getting a 'NO DATA' message when the connect routine does a 'test connect' & sends 'b0100' to get the supported PID's. Are you sure that your protocol is correct? You're manually sending protocol #1, which is 'SAE J1850 PWM'. Perhaps you need to double-check what protocol your vehicle uses.

In my case, the logic that deals with processing the 'supported PIDs' response message is broken. I'm using 'ISO 9141-2' protocol, and, even though the vehicle responds correctly with the supported PID's, the message is ignored & not processed by the pyOBD logic.

My Output
```
>>> c = obd.OBD("/dev/rfcomm0",protocol="3",baudrate=115200)
[obd.obd] ======================= python-OBD (v0.7.1) =======================
[obd.obd] Explicit port defined
[obd.elm327] Initializing ELM327: PORT=/dev/rfcomm0 BAUD=115200 PROTOCOL=3
[obd.elm327] write: b'ATZ\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'ELM327 v1.5\r\r>'
[obd.elm327] write: b'ATE0\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'ATE0\rOK\r\r>'
[obd.elm327] write: b'ATH1\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'ATL0\r'
[obd.elm327] wait: 1 seconds
[obd.elm327] read: b'OK\r\r>'
[obd.elm327] write: b'AT RV\r'
[obd.elm327] read: b'12.6V\r\r>'
[obd.elm327] write: b'ATTP3\r'
[obd.elm327] read: b'OK'
[obd.elm327] write: b'0100\r'
[obd.elm327] wait: 5 seconds
[obd.elm327] read: b'\r\r>48 6B 10 41 00 BE 1F B8 10 A9 \r\r>'
[obd.elm327] Connected Successfully: PORT=/dev/rfcomm0 BAUD=115200 PROTOCOL=3
[obd.obd] querying for supported commands
[obd.obd] Sending command: b'0100': Supported PIDs [01-20]
[obd.elm327] write: b'0100\r'
[obd.elm327] read: b'48 6B 10 41 00 BE 1F B8 10 A9 \r\r>'
[obd.OBDCommand] b'0100': Supported PIDs [01-20] did not receive any acceptable messages
[obd.obd] No valid data for PID listing command: b'0100': Supported PIDs [01-20]
[obd.obd] finished querying with 7 commands supported
[obd.obd] ===================================================================
```
FYI, I added additional delays in order to help deal with slow adapter & bluetooth lag.

JKondro, 2023-06-16:
I appreciate the reply, @csurf. You raise a good question regarding the protocol and I think that may be my mistake. I'm going to experiment with ISO 15765 this weekend on a '09 Ford Focus and a '08 Dodge Dakota and see what the debug output looks like.

At the moment I haven't added additional delays as I am using a corded ELM327 adapter. I might consider doing so too, however.

JKondro, 2023-06-17:
@csurf was right: I was trying for an OBD connection through the incorrect protocol. I was able to successful connect to a '09 Ford Focus and a '08 Dodge Dakota and read engine data using the ISO 15765-4 (CAN 11/500) protocol.

closed 2023-06-17 by JKondro