brendan-w/python-OBD #263: BUS INIT: STOPPED while trying to connect on Raspberry Pi

LightningBerk, 2023-06-26:
I cannot for the life of me figure out what is wrong. When trying to connect to the Bluetooth ELM327 adaptor on my 2002 Subaru Impreza WRX, I continuously get, read: b'BUS INIT: STOPPED\r\r>' when trying to connect which results with only 7 supported commands. I have scoured the internet and tried every fix I can and nothing has worked. I have made sure to use the correct protocol for my car and everything, any insight into what is going wrong would be great.

csurf, 2023-07-03:
My guess would be that you have a bad  elm327 clone that's having issues intializing its communications with the vehicle, and the 'BUS INIT' message is some weird, proprietary, undocumented response to this problem that's hard-coded into the firmware for your clone chip.

The reason I say this is because I have an OBD2 BT adapter with one of these chips (it's a YM1130 chip) that would occasionally put out strange, non-ELM327 compliant responses (something like b'VENDOR OPTION\r\r>' or b'VENDOR MODE\r\r>', can't remember).

hth

LightningBerk, 2023-07-03:
That's the consensus I was reaching based off of other threads. Do you have any recommendations for a reputable ELM327 bluetooth adaptor?

csurf, 2023-07-03:
@LightningBerk not at this moment. I'm in the same boat. Would like to get a better elm327 clone, as ELM went out of business, and the only other 'good' option is the STN series chips by OBD solutions, which support the ELM327 commands. However, all of the adapters I've seen that use those chips are way more expensive.

I've also seen adapters on aliexpress that claim to use actual PIC chips, so, in theory, they can be flashed with the original ELM327 firmware (version 1.4 I believe), but I honestly don't know if the old firmware will have weird bugs or issues dealing with the latest CAN stuff.

If you really need something legit, and don't mind paying more, I think the STN*-based adapters would be your best bet.

LightningBerk, 2023-07-04:
Gotcha, thanks for the info. You are correct, my car does not support CAN

csurf, 2023-07-04:
@LightningBerk If you're willing & able, it would be helpful if you could open up your adapter, take some pictures of the PCB, and post them here. It might help others who might have similar adapters.

LightningBerk, 2023-07-04:
Good idea, ill get to that later today