brendan-w/python-OBD #272: reading Odometer values

YoAzzi, 2024-03-23:
Hello 
is it possible to read the odometer value ? 
i have checked all the commande listed on the wiki but it does not contain it.
maybe the odometer is part of the Non-sensor data ?

ralam23, 2024-03-23:
hello! i'm a complete novice at this and new to coding. however it seems most ecus do support being able to pull mileage off the ecu. reading the documentation https://python-obd.readthedocs.io/en/latest/Custom%20Commands/ it seems you can make a custom command perhaps to pull mileage from the ecu! i'm not sure how to go about it but hopefully it's somewhat helpful :)

ralam23, 2024-03-23:
i stand mistaken. i asked chatgpt for the hex code for mileage info however it seems that doesnt exist. 

"Mileage data is often stored in the vehicle's EEPROM chip, not directly accessible via the diagnostic bus for security reasons. Some OBD readers might provide mileage information indirectly by calculating distance traveled since certain events, but this varies by reader and vehicle "

^ from chat gpt.

TastyHackTheBox, 2024-04-02:
Why do you trust ChatGPT so much? You know it lies all the time right?

2019+ vehicles were mandated in California to provide odometer the codes are: A6 166 - expect 4 bytes

Before 2019 its not so easy - relying on either avg speed * engine hours, or distance traveled before + after a CEL
Distance Since Reflash or Module Replacement - C7 199 - two bytes (yes this means only 65k KM)
Distance traveled since codes cleared - 31 49 - two bytes
Distance traveled with malfunction indicator lamp (MIL) on  - 21 33 - two bytes
Engine run time - 7F 127 - 13 bytes

Took me 4 minutes to google it: https://en.wikipedia.org/wiki/OBD-II_PIDs

YoAzzi, 2024-04-02:
the commands you have listed are car-dependent.
Thanks for the 4 minutes of research.