brendan-w/python-OBD #279: Vehicle automation question - workflow idea

zefoo, 2024-05-12:
Howdy. I spend a lot of time in my vehicle. Trying to think of creative things I can do with this library. I've already got it working with my bluetooth OBD reader and raspberry pi. 

One thing I was thinking. I'd love a "smart way" to track mileage for work related things. 

Success looks like: being able to get a text message after I drive, enter the purpose of the work trip, automatically get an excel file of start mileage, ending mileage, miles driven, and purpose of trip. 

Now, for how I'd do that with this library: I think I can pull odometer (still working on this, I have a 2020 vehicle, I should be able to create a custom command on this). I think I can poll for this, then I can poll for if vehicle is on. I can have some logic that says: "if vehicle is no longer on, then use the last recorded drive mileage, and ask me: "Was X mileage driven work related?" If I say "yes", then it asks me for the purpose and logs it.

I can figure out all the texting / excel / logging, etc, will likely even refine my workflow a bit so I can go in after the fact into a web UI or something. I want to make this as easy as humanly possible for me to track. 

I think I'd need to poll every minute, log odometer readings, log time. 

I'm still figuring out how to pull odometer reading and ideally I can pull lat/long of vehicle. 

Any other suggestions here? I think I've played with as many commands as I can, short any special commands for my vehicle. https://www.amazon.com/gp/product/B07JFRFJG6/ app seems to pull lat/long, so I think it's in there somewhere. 

Any other thoughts to add to this? Thought I'd ask since someone here may be more savvy, thank you!

thatlarrypearson, 2024-05-15:
The [SAE (Society of Automotive Engineers)](https://www.sae.org/standards/content/j1979-2_202104/) standard for OBD is SAE J1979/ISO 15031-5.  In the *Annex B - Parameter IDs* section there is ```Vehicle Odometer``` at ```PID``` ```0xA6``` that returns a value in tenths of kilometers.  I've only seen this PID available on my Ford vehicles from the 2019 and 2023 model years.

If you are interested in experimenting with different OBD capabilities, get a digital copy of the standard.   Not only will it help you identify useful OBD commands but it will help you understand underlying subtleties in how the commands actually work.

contrerasp, 2024-05-22:
This is essentially my end goal as well. I am currently using a project called [LubeLogger](https://github.com/hargata/lubelog) that works well. I have to type everything manually, but its not too bad. I would much prefer to leave a raspberry pi in my car though and have it send data back to a server once the pi is connected back to my home network. I will copy/paste the reply I found on [Issue#267](https://github.com/brendan-w/python-OBD/issues/267) with info regarding odometer.

What are you using to power the Pi in the car? I have heard before that cigarette lighter ports should not be used. I do not believe the USB ports on the bottom of the infotainment would be sufficient either. 

Per  @TastyHackTheBox ,
> 2019+ vehicles were mandated in California to provide odometer the codes are: A6 166 - expect 4 bytes You might get lucky if this was done previous to 2019 in your ECU.
> 
> Before 2019 its not so easy - relying on either engine hours, or distance traveled before + after a CEL as suggested above
> 
> Distance Since Reflash or Module Replacement - C7 199 - two bytes (yes this means only 65k KM) Distance traveled since codes cleared - 31 49 - two bytes Distance traveled with malfunction indicator lamp (MIL) on - 21 33 - two bytes Engine run time - 7F 127 - 13 bytes
> 
> Reference: https://en.wikipedia.org/wiki/OBD-II_PIDs
[Issue#267](https://github.com/brendan-w/python-OBD/issues/267)

thatlarrypearson, 2024-05-22:
Powering Raspberry Pi (RPi) 4 in vehicle is problematic in general.

I've had problems on my Fords (2017+) powering RPi connected to OBD via USB (wired) interface where a Ground Fault Protection device gets triggered and cuts power to the RPi.  This occurred when using 12VDC and 110VAC connections.  Solution is to switch to OBD.  This problem did not occur on my 2013 Jeep Wrangler.

Another power problem is RPi 4 requires 5 Volts at 3 Amps.  RPi 5 requires more.  Early on I burned out a few 12VDC/3A USB power adapters before switching exclusively to AC power supplies.  Also, not all USB cables will actually deliver 3 Amps without voltage drop.

All of my vehicles have factory support for 110VAC at 400 or more watts power.  This is more reliable so long as you don't plug in AC power until after the car has successfully started.  The RPi doesn't like the voltage drop from the starter motor at the beginning of the boot sequence.  I'm using "official" RPi power supplies.

I've put quite a bit of effort into getting my data collection software to reliably start automatically on boot on a headless Raspberry Pi.  You can find my engine data collection software on [thatlarrypearson on github](https://github.com/thatlarrypearson?tab=repositories) at [Telemetry OBD Logger README.md](https://github.com/thatlarrypearson/telemetry-obd).

The above repository provides detailed installation instructions that you can modify to fit your needs.  I wrote it and I find following it tedious.  I've been through several RPi OS upgrades and system builds/rebuilds and have kept the installation instructions up to date.  The system integration is tedious but necessary.  Lots of ```bash``` shell scripts augmented with Python.

I'm lucky - I've been working with UNIX/Linux since 1983.

contrerasp, 2024-05-23:
Wow, this project is nearly exactly what I've been looking to make for my Miata. Definitely going to have to check out this repo and see how you did a few things. Bashing my head into the wall trying to figure out why the has() function isn't returning data correctly. Im assuming its an user ID-10T error but who knows, I'm a silly guy who makes silly mistakes so that could also be it.