brendan-w/python-OBD #281: Import problem with Numpy2.0.1

Qooniee, 2024-08-13:
### Version Information
- Numpy version: 2.0.1
- Python version: 3.10.3
- python obd version: 0.7.2

### Problem Details
An error occurs when importing the obd module:
`AttributeError: module 'numpy' has no attribute 'cumproduct'
`

### Expected Behavior
The obd module should import without errors.

### Actual Behavior
An AttributeError occurs, indicating that the cumproduct function does not exist in the current version of numpy.

When I use numpy==1.26.4, I can import obd because numpy has np.cumproduct.
You can replace np.cumproduct with np.cumprod.

Lauwy222, 2024-08-20:
got the same issue

theaircrasher, 2024-09-03:
Same problem here with:

Python 3.10.12
numpy 2.1.0
obd 0.7.2

davrandom, 2024-11-15:
It is fixed in more recent versions of pint [1].
We just need to bump the requirements in python_OBD.

[1] https://pypi.org/project/Pint/