brendan-w/python-OBD #285: Upgrade dependency 'pint' from pint==0.20.* to pint==0.24.4 (Fedora 41, Python 3.13.1, python-obd 0.7.2)

DOSn3rd, 2025-01-03:
Installed python-ODB 0.7.2 from pip in Fedora 41 and got this error with pint 0.20.1:

```
Python 3.13.1 (main, Dec  9 2024, 00:00:00) [GCC 14.2.1 20240912 (Red Hat 14.2.1-3)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import obd
Traceback (most recent call last):
  File "<python-input-0>", line 1, in <module>
    import obd
  File "/home/dosn3rd/.local/lib/python3.13/site-packages/obd/__init__.py", line 41, in <module>
    from .obd import OBD
  File "/home/dosn3rd/.local/lib/python3.13/site-packages/obd/obd.py", line 38, in <module>
    from .commands import commands
  File "/home/dosn3rd/.local/lib/python3.13/site-packages/obd/commands.py", line 36, in <module>
    from .decoders import *
  File "/home/dosn3rd/.local/lib/python3.13/site-packages/obd/decoders.py", line 38, in <module>
    from .UnitsAndScaling import Unit, UAS_IDS
  File "/home/dosn3rd/.local/lib/python3.13/site-packages/obd/UnitsAndScaling.py", line 38, in <module>
    Unit = pint.UnitRegistry()
  File "/home/dosn3rd/.local/lib/python3.13/site-packages/pint/facets/plain/registry.py", line 135, in __call__
    obj = super().__call__(*args, **kwargs)
  File "/home/dosn3rd/.local/lib/python3.13/site-packages/pint/registry.py", line 92, in __init__
    super().__init__(
    ~~~~~~~~~~~~~~~~^
        filename=filename,
        ^^^^^^^^^^^^^^^^^^
    ...<11 lines>...
        cache_folder=cache_folder,
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/home/dosn3rd/.local/lib/python3.13/site-packages/pint/facets/system/registry.py", line 52, in __init__
    super().__init__(**kwargs)
    ~~~~~~~~~~~~~~~~^^^^^^^^^^
  File "/home/dosn3rd/.local/lib/python3.13/site-packages/pint/facets/group/registry.py", line 40, in __init__
    super().__init__(**kwargs)
    ~~~~~~~~~~~~~~~~^^^^^^^^^^
  File "/home/dosn3rd/.local/lib/python3.13/site-packages/pint/facets/context/registry.py", line 65, in __init__
    super().__init__(**kwargs)
    ~~~~~~~~~~~~~~~~^^^^^^^^^^
  File "/home/dosn3rd/.local/lib/python3.13/site-packages/pint/facets/nonmultiplicative/registry.py", line 46, in __init__
    super().__init__(**kwargs)
    ~~~~~~~~~~~~~~~~^^^^^^^^^^
  File "/home/dosn3rd/.local/lib/python3.13/site-packages/pint/facets/plain/registry.py", line 217, in __init__
    from ... import delegates  # TODO: change thiss
    ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dosn3rd/.local/lib/python3.13/site-packages/pint/delegates/__init__.py", line 11, in <module>
    from . import txt_defparser
  File "/home/dosn3rd/.local/lib/python3.13/site-packages/pint/delegates/txt_defparser/__init__.py", line 12, in <module>
    from .defparser import DefParser
  File "/home/dosn3rd/.local/lib/python3.13/site-packages/pint/delegates/txt_defparser/defparser.py", line 9, in <module>
    from . import block, common, context, defaults, group, plain, system
  File "/home/dosn3rd/.local/lib/python3.13/site-packages/pint/delegates/txt_defparser/common.py", line 21, in <module>
    @dataclass(frozen=True)
     ~~~~~~~~~^^^^^^^^^^^^^
  File "/usr/lib64/python3.13/dataclasses.py", line 1295, in wrap
    return _process_class(cls, init, repr, eq, order, unsafe_hash,
                          frozen, match_args, kw_only, slots,
                          weakref_slot)
  File "/usr/lib64/python3.13/dataclasses.py", line 1043, in _process_class
    raise TypeError('cannot inherit frozen dataclass from a '
                    'non-frozen one')
TypeError: cannot inherit frozen dataclass from a non-frozen one
>>> 
```

Seems to be related to a few errors in the dependency "pint" that may even have come from issues within cpython itself?
https://github.com/hgrecco/pint/issues?q=is%3Aissue+pint%2Fdelegates%2Ftxt_defparser%2Fcommon.py

Fix/Workaround:
@hgrecco seems to have fixed this or similar issues in pint==0.24.4, and as a workaround i installed that version even if pip complains about unsatisfied dependencies.
https://github.com/hgrecco/pint/issues/2074#issuecomment-2462693795

With pint upgraded to 0.24.4 the issue is no longer present, at least for me.
```
Python 3.13.1 (main, Dec  9 2024, 00:00:00) [GCC 14.2.1 20240912 (Red Hat 14.2.1-3)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import obd
>>> 
```

I have not tested this with obdsim but I tried the workaround by connecting to a car, and I can successfully see that i have no DTCs and reading Mode 1 "Live Data" seems to work as intended with pint upgraded to version 0.24.4.

imtraviz, 2025-02-07:
``I am getting this same issue 

`
python -m venv venv
source venv/bin/activate
pip install obd pandas
`

fixed first this 
`AttributeError: module 'numpy' has no attribute 'cumproduct'. Did you mean: 'cumprod'?`

by changing `for func_str in ["cumprod","cumproduct", "nancumprod"]:`
`for func_str in ["cumprod", "nancumprod"]:` on line 884

but now i am getting 
`File "/usr/lib/python3.13/dataclasses.py", line 1295, in wrap
    return _process_class(cls, init, repr, eq, order, unsafe_hash,
                          frozen, match_args, kw_only, slots,
                          weakref_slot)
  File "/usr/lib/python3.13/dataclasses.py", line 1043, in _process_class
    raise TypeError('cannot inherit frozen dataclass from a '
                    'non-frozen one')
TypeError: cannot inherit frozen dataclass from a non-frozen one
`

so how to fix this

travis2319, 2025-02-08:
it works on lower python version like 3.11