brendan-w/python-OBD #297: No available ports

gordona23, 2025-06-02:
Trying to run Python-OBD on Raspberry Pi Zero 2W.  Here's the code:

import sys
sys.path.insert(0, '/home/gordona/my-venv/lib/python3.11/site-packages')
import obd
obd.logger.setLevel(obd.logging.DEBUG)
ports = obd.scan_serial()
print (ports)
connection = obd.OBD()
fast=False
timeout=30
cmd = obd.commands.HYBRID_BATTERY_REMAINING
response = connection.query(cmd)
print (response.value)

Here's the output:

[]
[obd.obd] ======================= python-OBD (v0.7.3) =======================
[obd.obd] Using scan_serial to select port
[obd.obd] Available ports: []
[obd.obd] No OBD-II adapters found
[obd.obd] Cannot load commands: No connection to car
[obd.obd] ===================================================================
[obd.obd] Query failed, no connection available

It looks like we're not finding any ports for the BLE connection?  Any suggestions would be appreciated.   Thx

barracuda-fsh, 2025-06-03:
I think you need to use these two commands, then restart and it will find ports.

gordona23, 2025-06-03:
Thank your for the prompt response.  Both commands ran successfully but no change in the out# ##put of my program. It still looks like obd.scan_serial is not finding any usable ports.   I am getting some warnings that may be germaine.  I apologize for not including them in the original post. 

WARNING: Your `/home/gordona/my-venv/lib/python3.11/site-packages/serial/__init__.py`
 is shadowing the library module 'serial'. Consider renaming or moving it!

WARNING: Your `/home/gordona/my-venv/lib/python3.11/site-packages/typing_extensions.py
` is shadowing the library module 'typing_extensions'. Consider renaming or moving it!

WARNING: Your `/home/gordona/my-venv/lib/python3.11/site-packages/platformdirs
/__init__.py` is shadowing the library module 'platformdirs'. Consider renaming 
or moving it!

gordona23, 2025-06-04:
More information:  I ran blue-serial and it was able to find the OBD device.  Here's the output from the run:

Finished general BLE scan
(my-venv) gordona@OBD:~/my-venv $ ble-serial -d 66:1e:87:01:93:ef
17:35:12.839 | INFO | linux_pty.py: Port endpoint created on /tmp/ttyBLE -> /dev/pts/1
17:35:12.840 | INFO | ble_client.py: Receiver set up
17:35:13.690 | INFO | ble_client.py: Trying to connect with 66:1E:87:01:93:EF: VEEPEAK
17:35:15.059 | INFO | ble_client.py: Device 66:1E:87:01:93:EF connected
17:35:15.061 | INFO | ble_client.py: Found write-without-response characteristic 0000fff2-0000-1000-8000-00805f9b34fb (H. 15)
17:35:15.063 | INFO | ble_client.py: Found notify characteristic 0000fff1-0000-1000-8000-00805f9b34fb (H. 17)
17:35:15.569 | INFO | main.py: Running main loop!

Does that help explain why obd.scan+serial() is not finding any ports ?

Thanks in advance for any insights

gordona23, 2025-06-07:
I just ran bye_serial, in my virtual environment, as part of a Python script and it found the ports and the OBD device.  So it's able to run in the same environment that Python-OBD is having problems with.   Any thoughts ??